reynico/raspberry-noaa #84: Scheduler not generating passes.

Anjum9694, 2020-12-21:
Hi I installed using the installer script and rebooted the Pi. The cornjob doesn't seem to be working so I ran the bash command verbose, don't see any errors but I ran atq afterwards there doesn't seem to be any passes. The webserver works fine but it doesn't show any passes either.

[image]

reynico, 2020-12-21:
Hi!
Most likely that's because there isn't any new pass for this day. Judging by your Github profile you're located in Maldives (GMT +5) so it's almost midnight. The cronjob will run automatically in a few minutes (00:01)!

Anjum9694, 2020-12-21:
LOL yeah. Glad there's no issue with my installation then.
[image]

closed 2020-12-21 by Anjum9694