reynico/raspberry-noaa #85: Problems wish image displaying.

tegwilym, 2020-12-22:
I installed the latest build yesterday. The new web interface looks great!  
My problem seem to be file and/or folder permissions on the /var/www/wx/images folder.  
I get the web page fine, but there are no .jpg images in the folder they should be. I tried changing the folder ownership to pi:pi which did allow the thumb folder creation, but no images.  (I did try a reinstall of the software from the auto installer, but still the same issue). 
It is receiving the images fine since I do have proper .wave files and manually processing in xwxtoimg I get good images (no map overlay though). 
I've used the previous version that installed and ran perfectly with  no problems with the install.sh script. 
Here is a log from the 'pi' file in the /var/mail folder if this helps. 

`From pi@raspberrypi Mon Dec 21 10:35:59 2020
Return-path: <pi@raspberrypi>
Envelope-to: pi@raspberrypi
Delivery-date: Mon, 21 Dec 2020 10:35:59 -0600
Received: from pi by raspberrypi with local (Exim 4.92)
        (envelope-from <pi@raspberrypi>)
        id 1krOAB-0001Mc-Ff
        for pi@raspberrypi; Mon, 21 Dec 2020 10:35:59 -0600
Subject: Output from your job       30
To: pi@raspberrypi
Message-Id: <E1krOAB-0001Mc-Ff@raspberrypi>
From: pi@raspberrypi
Date: Mon, 21 Dec 2020 10:35:59 -0600

INFO : Starting rtl_fm record
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 49.60 dB.
activated bias-T on GPIO PIN 0
Tuned to 138392500 Hz.
Oversampling input by: 32x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Sampling at 1920000 S/s.
Output at 60000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Bulding pass map
Satellite: NOAA 18
Pass Start: 2020-12-21 16:19:53 UTC
Pass Duration: 15:27
Elevation: 56
Azimuth: 290
Direction: southbound
..INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201221-101953-ZA.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201221-101953-ZA.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201221-101953-ZA.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201221-101953-ZA.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201221-101953-ZA.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201221-101953-MCIR.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201221-101953-MCIR.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201221-101953-MCIR.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201221-101953-MCIR.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201221-101953-MCIR.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201221-101953-MCIR-precip.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201221-101953-MCIR-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201221-101953-MCIR-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201221-101953-MCIR-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201221-101953-MCIR-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Moving audio files out to the SD card
`

reynico, 2020-12-22:
Hi!
Could you please show me the output of
```
ls -al /var/www/wx/
```

tegwilym, 2020-12-22:
> Hi!
> Could you please show me the output of
> 
> ```
> ls -al /var/www/wx/
> ```

pi@raspberrypi:~ $ ls -al /var/www/wx
total 84
drwxrwxr-x 9 www-data www-data  4096 Dec 21 18:08 .
drwxr-xr-x 4 root     root      4096 Dec 20 20:12 ..
drwxrwxr-x 2 www-data www-data  4096 Dec 21 22:03 audio
-rw-r--r-- 1 pi       pi         104 Dec 21 18:08 Config.php
drwxr-xr-x 2 pi       pi        4096 Dec 20 20:06 Controller
-rw-r--r-- 1 pi       pi         206 Dec 20 20:06 detail.php
-rw-r--r-- 1 pi       pi         146 Dec 20 20:06 footer.php
-rw-r--r-- 1 pi       pi         976 Dec 20 20:06 header.php
drwxr-xr-x 3 www-data www-data  4096 Dec 20 23:14 images
-rw-r--r-- 1 pi       pi         185 Dec 20 20:06 index.php
drwxr-xr-x 2 pi       pi        4096 Dec 20 20:06 language
-rwxr-xr-x 1 pi       pi       16850 Dec 20 20:06 logo-small.png
drwxr-xr-x 3 www-data www-data  4096 Dec 20 20:16 meteor
drwxr-xr-x 2 pi       pi        4096 Dec 20 20:06 Model
-rw-r--r-- 1 pi       pi         108 Dec 20 20:06 passes.php
-rw-r--r-- 1 pi       pi        1509 Dec 20 20:06 style.css
drwxr-xr-x 2 pi       pi        4096 Dec 20 20:06 Views

reynico, 2020-12-22:
Uhm, seems like the `/var/www/wx/` folder has the wrong permissions.

```
sudo chown pi:pi /var/www/wx
```

that should be enough

closed 2020-12-27 by reynico