reynico/raspberry-noaa #86: No images from passes.

Anjum9694, 2020-12-22:
Hi so the passes were generated by scheduler, I left it overnight but next day there were no images from the passes. I tested with rtl_test to be sure my v3 was working with the pi. What could be the reason for it?

[image]
[image]

reynico, 2020-12-22:
seems like the `/var/www/wx/` folder has the wrong permissions.

```
sudo chown pi:pi /var/www/wx
```

that should be enough

Anjum9694, 2020-12-23:
I applied the permission fixes. I will be submitting here the changes I am doing.

I noticed the schedule script doesn't update the tle database of predict, it's way off! 
[image]
[image]

I temporarily moved the weather.tle from raspberry-noaa predict folder as predict.tle. Now I am doing a visible pass calculation.
[image]

The timing is off by about 5 hours. The date command returns the correct timezone. I'm not sure where needs to be changed. That's all for now.

reynico, 2020-12-23:
Hi, as far I know, predict uses a different path for its database. raspberry-noaa uses a custom path for it.

Anjum9694, 2020-12-24:
[image]
I did a reinstall. Got a standard in syntax error there. My timezone is GMT +5 so I wrote +5

reynico, 2020-12-24:
Hi, for positive timezones it is just the number (i.e. 5)

Anjum9694, 2020-12-24:
Thanks! I am getting accurate passes now, which I hope will result in pictures finally!
Just one more thing, sometimes when the scheduler is run I get 
date: invalid date ‘@’
date: invalid date ‘@’
date: invalid date ‘@’
date: invalid date ‘@’

So when crontab is run and the error occurs there are no passes for the day.

reynico, 2021-01-03:
Hi! Sorry for the delay, I completely forgot this issue. Have you figured out your problem?

dom1706, 2021-01-04:
Hello and I would like to thank reynico for his great job !
I've the same problem: no pictures but the other page with passes is ok.
I've this error message: 
`convert: unable to open image `/var/www/wx/images/NOAA1520210104-091024-MCIR-precip.jpg': Aucun fichier ou dossier de ce type @ error/blob.c/OpenBlob/2874.`
I have put this to give the right permissions but it doesn't help: sudo chown pi:pi /var/www/wx
What could be my problem ?
Best regards.

Anjum9694, 2021-01-04:
Hey Nico, I have uninstalled it for now. Since it's working for others I figured it's best to reimage my sd card and start over. Will update this thread to let you know how it goes. Best regards and a Happy New Year.

surfrod, 2021-01-05:
ok, against my better judgement ( 🥇 ) I did a fresh pull and restore.

But now, not getting any images and see error messages galore like:
INFO : Bulding pass map
Satellite: NOAA 19
Pass Start: 2021-01-05 12:06:46 UTC
Pass Duration: 15:34
Elevation: 62
Azimuth: 288
Direction: southbound
..INFO : Decoding image
Satellite: NOAA 19
Status: signal processing...
wxtoimg: error: out of memory in line 1412 of noaa.c
wxtoimg: warning: enhancement ignored: could not find a NOAA/GOES sensor 4 image
INFO : Decoding image

I have 54Gig free disk memory, and /var/www/wx permissions are fine.  .noaa.conf looks good.

Just did a mv raspberry-noaa to OLDraspberry-noaa and did a new git pull; cd; ./install.sh

So, what do I have to fix/check/redo to get back up again ?

surfrod, 2021-01-05:
Meteor decode error:
INFO : Starting rtl_fm record
/usr/local/bin/rtl_fm: option requires an argument -- 'g'
rtl_fm, a simple narrow band FM demodulator for RTL2832 based DVB-T receivers

Use:	rtl_fm -f freq [-options] [filename]
	-f frequency_to_tune_to [Hz]
	    use multiple -f for scanning (requires squelch)
	    ranges supported, -f 118M:137M:25k
	[-M modulation (default: fm)]
	    fm, wbfm, raw, am, usb, lsb
	    wbfm == -M fm -s 170k -o 4 -A fast -r 32k -l 0 -E deemp
	    raw mode outputs 2x16 bit IQ pairs
	[-s sample_rate (default: 24k)]

surfrod, 2021-01-05:
another error:
INFO : Bulding pass map
wxmap: error: /home/pi/raspberry-noaa/receive.sh: line 45:  6284 Segmentation fa
ult      /usr/local/bin/wxmap -T "${1}" -H "${4}" -p 0 -l 0 -o "${PASS_START}" "
${NOAA_HOME}/map/${3}-map.png"
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOA
A1820210104-214101-map.png'
Satellite: other
Status: signal processing...
wxtoimg: warning: enhancement ignored (ZA) because satellite is not NOAA or GOES
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOA
A1820210104-214101

surfrod, 2021-01-05:
btw I'm running on a pi4

disk partitions:
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        63G  8.7G   52G  15% /
devtmpfs        2.0G     0  2.0G   0% /dev
tmpfs           2.1G  4.1k  2.1G   1% /dev/shm
tmpfs           2.1G  9.0M  2.0G   1% /run
tmpfs           5.3M  4.1k  5.3M   1% /run/lock
tmpfs           2.1G     0  2.1G   0% /sys/fs/cgroup
tmpfs           1.1G     0  1.1G   0% /var/ramfs
/dev/mmcblk0p1  265M   56M  209M  22% /boot
tmpfs           402M  4.1k  402M   1% /run/user/1000

surfrod, 2021-01-05:
ok, searched for "GAIN" in the issues and realized that ./install.sh did not overwrite $HOME/.noaa.conf

Just copied that and will see what happens the next pass(es)!

reynico, 2021-01-05:
hi!

> wxtoimg: error: out of memory in line 1412 of noaa.c

This is an annoying error, got it a few times but I think it's not related to anything about the hardware, just a bug in xwxtoimg. and since we don't have the source code it's pretty tricky to find out what's going on inside.

> ok, searched for "GAIN" in the issues and realized that ./install.sh did not overwrite $HOME/.noaa.conf

good to hear! I'm including migrations for new improvements so users don't hit the wall with misconfigs

Piccar, 2021-01-06:
> Thanks! I am getting accurate passes now, which I hope will result in pictures finally!
> Just one more thing, sometimes when the scheduler is run I get
> date: invalid date ‘@’
> date: invalid date ‘@’ [...]

Noticed that this only happens to me when my raspi has no Internet connection.

reynico, 2021-01-08:
Hi @Piccar !

Mmm, that seems to be done because of the TLE file overwrite on `schedule.sh`.

Comment out or delete those lines on `schedule.sh`
```
wget -qr http://www.celestrak.com/NORAD/elements/weather.txt -O "${NOAA_HOME}"/predict/weather.txt
wget -qr http://www.celestrak.com/NORAD/elements/amateur.txt -O "${NOAA_HOME}"/predict/amateur.txt
```

then, with an internet connection available:

```
cd /home/pi/raspberry/noaa/predict/
wget -qr http://www.celestrak.com/NORAD/elements/weather.txt
wget -qr http://www.celestrak.com/NORAD/elements/amateur.txt
```

Anjum9694, 2021-01-08:
>wget -qr http://www.celestrak.com/NORAD/elements/weather.txt -O "${NOAA_HOME}"/predict/weather.txt
>wget -qr http://www.celestrak.com/NORAD/elements/amateur.txt -O "${NOAA_HOME}"/predict/amateur.txt

Maybe you can have schedule.sh grab the new TLE after a quick ping test for internet connection? If no connection use existing TLE files.

Anjum9694, 2021-01-25:
Reinstalled this now on a fresh Pi installation and since I'm technically getting images now you may close this issue.
I wrote this on schedule.sh so the process doesn't stop if the internet is down.

```
if [[ "$(ping -c 1 8.8.8.8 | grep '100% packet loss' )" != "" ]]; then
    echo "INTERNET IS DOWN, SKIPPING TLE UPDATE"
    exit 1
else
    echo "INTERNET IS UP, UPDATING TLE"
	wget -qr http://www.celestrak.com/NORAD/elements/weather.txt -O "${NOAA_HOME}"/predict/weather.txt
	wget -qr http://www.celestrak.com/NORAD/elements/amateur.txt -O "${NOAA_HOME}"/predict/amateur.txt
    exit 0
fi
```

Anjum9694, 2021-01-25:
> Noticed that this only happens to me when my raspi has no Internet connection.

The reason this happens is because wget writes an empty file when it fails to download the tle file, so the predictor software gets confused. I commented out the tle update part of the scheduler.sh and wrote my own separate 100% failproof script for the job. It makes sure the downloaded tle file has the right size before stopping. I am closing this thread as there is no problem now.

```
#!/bin/sh
cd /home/pi/raspberry-noaa/predict/
curl --fail http://www.celestrak.com/NORAD/elements/weather.txt -o /home/pi/raspberry-noaa/predict/weather.txt
curl --fail http://www.celestrak.com/NORAD/elements/amateur.txt -o /home/pi/raspberry-noaa/predict/amateur.txt
until [[ "$(stat -c '%s' /home/pi/raspberry-noaa/predict/weather.txt | grep '8904' )" != "" ]]; do
    curl --fail http://www.celestrak.com/NORAD/elements/weather.txt -o /home/pi/raspberry-noaa/predict/weather.txt
done
until [[ "$(stat -c '%s' /home/pi/raspberry-noaa/predict/amateur.txt | grep '13776' )" != "" ]]; do
    curl --fail http://www.celestrak.com/NORAD/elements/amateur.txt -o /home/pi/raspberry-noaa/predict/amateur.txt
done

echo TLE HAS BEEN UPDATED.
```

closed 2021-01-25 by Anjum9694