reynico/raspberry-noaa #87: [Error] NOAA and Meteor-M2 images not getting decoded

mihajlo2003petkovic, 2020-12-22:
Hi! I've had issue with the newest version that NOAA and Meteor-M2 images are not getting decoded. This is the latest log for NOAA satellites:

```
From pi@raspberrypi Tue Dec 22 08:55:14 2020
Return-path: <pi@raspberrypi>
Envelope-to: pi@raspberrypi
Delivery-date: Tue, 22 Dec 2020 08:55:14 +0100
Received: from pi by raspberrypi with local (Exim 4.92)
        (envelope-from <pi@raspberrypi>)
        id 1krcVm-0000mB-N1
        for pi@raspberrypi; Tue, 22 Dec 2020 08:55:14 +0100
Subject: Output from your job        8
To: pi@raspberrypi
Message-Id: <E1krcVm-0000mB-N1@raspberrypi>
From: pi@raspberrypi
Date: Tue, 22 Dec 2020 08:55:14 +0100

INFO : Starting rtl_fm record
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 7.70 dB.
Tuned to 138392500 Hz.
Oversampling input by: 32x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Sampling at 1920000 S/s.
Output at 60000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Bulding pass map
Satellite: NOAA 18
Pass Start: 2020-12-22 07:39:31 UTC
Pass Duration: 14:52
Elevation: 31
Azimuth: 96
Direction: southbound
..INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201222-083930-MSA-precip.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-MSA-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201222-083930-MSA-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-MSA-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201222-083930-MSA-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201222-083930-MSA.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-MSA.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201222-083930-MSA.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-MSA.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201222-083930-MSA.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201222-083930-MCIR-precip.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-MCIR-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201222-083930-MCIR-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-MCIR-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201222-083930-MCIR-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201222-083930-HVCT-precip.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-HVCT-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201222-083930-HVCT-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-HVCT-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201222-083930-HVCT-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201222-083930-HVCT.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-HVCT.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201222-083930-HVCT.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-HVCT.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201222-083930-HVCT.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201222-083930-therm.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-therm.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201222-083930-therm.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-therm.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201222-083930-therm.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201222-083930-sea-day.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-sea-day.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201222-083930-sea-day.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-sea-day.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201222-083930-sea-day.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1820201222-083930-histeq.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-histeq.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1820201222-083930-histeq.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1820201222-083930-histeq.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1820201222-083930-histeq.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Posting to Twitter
Traceback (most recent call last):
  File "/home/pi/raspberry-noaa/post.py", line 23, in <module>
    res = api.media_upload(filename)
  File "/usr/local/lib/python3.7/dist-packages/tweepy/api.py", line 223, in media_upload
    file_type = imghdr.what(filename)
  File "/usr/lib/python3.7/imghdr.py", line 16, in what
    f = open(file, 'rb')
FileNotFoundError: [Errno 2] No such file or directory: '/var/www/wx/images/NOAA1820201222-083930-MSA-precip.jpg'
INFO : Deleting audio files
```
And this is log for Meteor M2. It is longer than usual since I've added piece of code that creates additional images (true-colour, false-colour and infrared), but I've also tried the original code and it didn't work. This is the [Meteor Log.txt](https://github.com/reynico/raspberry-noaa/files/5728884/Meteor.Log.txt). I've added it as a separate file since it is too big to display on screen.

This is my `receive_meteor.sh` script. The problem is that I cannot find any images decoded in `/var/www/wx/images` or elsewhere.

```
pi@raspberrypi:~/raspberry-noaa $ cd /var/www/wx/images
pi@raspberrypi:/var/www/wx/images $ ls
thumb
pi@raspberrypi:/var/www/wx/images $ cd thumb
pi@raspberrypi:/var/www/wx/images/thumb $ ls
pi@raspberrypi:/var/www/wx/images/thumb $
```

reynico, 2020-12-22:
seems like the `/var/www/wx/` folder has the wrong permissions.

```
sudo chown pi:pi /var/www/wx
```

that should be enough

mihajlo2003petkovic, 2020-12-22:
I've changed permissions, will tell you later how it goes. You should include that in your `install.sh` script. Yesterday I had to manually create `/var/www/wx/images` directory using `sudo mkdir /var/www/wx/images`. I suspect that's due to the wrong permissions.

reynico, 2020-12-22:
yeah, i think i did a regression on one of the last commits

mihajlo2003petkovic, 2020-12-22:
I'm still not getting any images. This is the latest NOAA log:

```
From pi@raspberrypi Tue Dec 22 17:55:54 2020
Return-path: <pi@raspberrypi>
Envelope-to: pi@raspberrypi
Delivery-date: Tue, 22 Dec 2020 17:55:54 +0100
Received: from pi by raspberrypi with local (Exim 4.92)
        (envelope-from <pi@raspberrypi>)
        id 1krkx0-0000f7-Ng
        for pi@raspberrypi; Tue, 22 Dec 2020 17:55:54 +0100
Subject: Output from your job       13
To: pi@raspberrypi
Message-Id: <E1krkx0-0000f7-Ng@raspberrypi>
From: pi@raspberrypi
Date: Tue, 22 Dec 2020 17:55:54 +0100

INFO : Starting rtl_fm record
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 7.70 dB.
Tuned to 138100000 Hz.
Oversampling input by: 32x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Sampling at 1920000 S/s.
Output at 60000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Bulding pass map
Satellite: NOAA 15
Pass Start: 2020-12-22 16:40:14 UTC
Pass Duration: 14:46
Elevation: 48
Azimuth: 70
Direction: northbound
..INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1520201222-174013-MCIR-precip.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-MCIR-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1520201222-174013-MCIR-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-MCIR-precip.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1520201222-174013-MCIR-precip.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1520201222-174013-MCIR.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-MCIR.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1520201222-174013-MCIR.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-MCIR.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1520201222-174013-MCIR.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1520201222-174013-therm.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-therm.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1520201222-174013-therm.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-therm.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1520201222-174013-therm.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1520201222-174013-sea.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-sea.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1520201222-174013-sea.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-sea.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1520201222-174013-sea.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Decoding image
wxtoimg: error: could not open output file /var/www/wx/images/NOAA1520201222-174013-histeq.jpg: Permission denied
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-histeq.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/NOAA1520201222-174013-histeq.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/var/www/wx/images/NOAA1520201222-174013-histeq.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/var/www/wx/images/thumb/NOAA1520201222-174013-histeq.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Posting to Twitter
Traceback (most recent call last):
  File "/home/pi/raspberry-noaa/post.py", line 23, in <module>
    res = api.media_upload(filename)
  File "/usr/local/lib/python3.7/dist-packages/tweepy/api.py", line 223, in media_upload
    file_type = imghdr.what(filename)
  File "/usr/lib/python3.7/imghdr.py", line 16, in what
    f = open(file, 'rb')
FileNotFoundError: [Errno 2] No such file or directory: '{NOAA_OUTPUT}/images/NOAA1520201222-174013-MCIR-precip.jpg'
INFO : Deleting audio files
```

mihajlo2003petkovic, 2020-12-22:
```
pi@raspberrypi:~ $ cd /var/www/wx
pi@raspberrypi:/var/www/wx $
pi@raspberrypi:/var/www/wx $ ls -lH
total 76
drwxrwxr-x 2 pi       pi        4096 Dec 21 20:04 audio
-rw-r--r-- 1 pi       pi         104 Dec 21 21:30 Config.php
drwxr-xr-x 2 pi       pi        4096 Dec 21 21:24 Controller
-rw-r--r-- 1 pi       pi         206 Dec 21 19:41 detail.php
-rw-r--r-- 1 pi       pi         146 Dec 21 19:41 footer.php
-rw-r--r-- 1 pi       pi         995 Dec 21 21:24 header.php
drwxr-xr-x 3 www-data www-data  4096 Dec 21 20:05 images
-rw-r--r-- 1 pi       pi         185 Dec 21 19:41 index.php
drwxr-xr-x 2 pi       pi        4096 Dec 21 19:41 language
-rwxr-xr-x 1 pi       pi       16850 Dec 21 19:41 logo-small.png
drwxr-xr-x 3 pi       pi        4096 Dec 22 18:50 meteor
drwxr-xr-x 2 pi       pi        4096 Dec 21 21:23 Model
-rw-r--r-- 1 pi       pi         108 Dec 21 19:41 passes.php
-rw-r--r-- 1 pi       pi        1509 Dec 21 19:41 style.css
drwxr-xr-x 2 pi       pi        4096 Dec 21 21:23 Views

```

reynico, 2020-12-22:
Try changing the ownership of images:

```
sudo chown pi:pi /var/www/wx/images
sudo chown -R pi:pi /var/www/wx/images

```

tegwilym, 2020-12-22:
I'll try the same changes on mine too. Seems like a similar problem, I'm not getting NOAA or Meteor.  Just  placeholder for images, but nothing in the image folders.   I do get .wav files so it's working just can't save the processed images.

tegwilym, 2020-12-22:
Should the meteor folder also be pi:pi?  This is what mine currently looks like: 

`pi@raspberrypi:/var/www/wx $ ls -la
total 84
drwxrwxr-x 9 pi       pi        4096 Dec 21 18:08 .
drwxr-xr-x 4 root     root      4096 Dec 20 20:12 ..
drwxrwxr-x 2 www-data www-data  4096 Dec 22 10:24 audio
-rw-r--r-- 1 pi       pi         104 Dec 21 18:08 Config.php
drwxr-xr-x 2 pi       pi        4096 Dec 20 20:06 Controller
-rw-r--r-- 1 pi       pi         206 Dec 20 20:06 detail.php
-rw-r--r-- 1 pi       pi         146 Dec 20 20:06 footer.php
-rw-r--r-- 1 pi       pi         976 Dec 20 20:06 header.php
drwxr-xr-x 3 pi       pi        4096 Dec 20 23:14 images
-rw-r--r-- 1 pi       pi         185 Dec 20 20:06 index.php
drwxr-xr-x 2 pi       pi        4096 Dec 20 20:06 language
-rwxr-xr-x 1 pi       pi       16850 Dec 20 20:06 logo-small.png
drwxr-xr-x 3 www-data www-data  4096 Dec 20 20:16 meteor
drwxr-xr-x 2 pi       pi        4096 Dec 20 20:06 Model
-rw-r--r-- 1 pi       pi         108 Dec 20 20:06 passes.php
-rw-r--r-- 1 pi       pi        1509 Dec 20 20:06 style.css
drwxr-xr-x 2 pi       pi        4096 Dec 20 20:06 Views
`

reynico, 2020-12-22:
yeah, do the same for audio and meteor folders. If this manual fix works for you guys I'll update the install script.

tegwilym, 2020-12-22:
> yeah, do the same for audio and meteor folders. If this manual fix works for you guys I'll update the install script.

Cool!  
One bonehead question, since this was added to the repository, can I do an 'apt-get upgrade' or 'git pull' command when things are changed?  I've messed things up in the past and just ended up doing a full re-install, but that's not very efficient!  :-)

reynico, 2020-12-22:
Yeah, just 
```
git fetch -pt && git pull
```
for the changes inside the raspberry-noaa folder.

mihajlo2003petkovic, 2020-12-22:
> Try changing the ownership of images:
> 
> ```
> sudo chown pi:pi /var/www/wx/images
> sudo chown -R pi:pi /var/www/wx/images
> ```

This worked for me! Thanks!
As a suggestion: write a simple for loop that changes ownership for every directory in `/var/www/wx` to pi.

mihajlo2003petkovic, 2020-12-22:
Maybe this?

```
for i in /var/www/wx/*; do
    if [ -d "$i" ]; then
        sudo chown pi:pi  $i
        sudo chown -R pi:pi $i
    fi
done
```

tegwilym, 2020-12-22:
I tried all the changes, mentioned.  Just waiting now.....
Thanks for the clarification with the updating, that will save me a lot of time!

mihajlo2003petkovic, 2020-12-23:
This is new Meteor-M2 log. I'm still not getting any images out of it. It was a high pass, 59 degrees.

```
From pi@raspberrypi Wed Dec 23 18:51:46 2020
Return-path: <pi@raspberrypi>
Envelope-to: pi@raspberrypi
Delivery-date: Wed, 23 Dec 2020 18:51:46 +0100
Received: from pi by raspberrypi with local (Exim 4.92)
        (envelope-from <pi@raspberrypi>)
        id 1ks8Ic-0000en-0E
        for pi@raspberrypi; Wed, 23 Dec 2020 18:51:46 +0100
Subject: Output from your job       16
To: pi@raspberrypi
Message-Id: <E1ks8Ic-0000en-0E@raspberrypi>
From: pi@raspberrypi
Date: Wed, 23 Dec 2020 18:51:46 +0100

INFO : The system doesn't have enough space to store a Meteor pass on RAM
INFO : I'll flip this image pass because FLIP_METEOR_IMG is set to true
INFO : Starting rtl_fm record
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 7.70 dB.
Tuned to 137388000 Hz.
Oversampling input by: 4x.
Oversampling output by: 1x.
Buffer size: 7.11ms
Sampling at 1152000 S/s.
Output at 288000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Demodulation in progress (QPSK)

Meteor-M2 LRPT demodulator v0.3-beta1

(18:30:50) Input: /var/www/wx/meteor/audio/METEOR-M220201223-181535.wav, output: /var/www/wx/meteor/METEOR-M220201223-181535.qpsk
(18:30:50) Input samplerate: 96000
(18:30:50) Demodulator initialized
(18:30:50) (  0.0%) Carrier:   +11.5 Hz, Locked: No
(18:30:55) (  0.8%) Carrier:  +551.8 Hz, Locked: No
(18:31:00) (  1.8%) Carrier: -1145.4 Hz, Locked: No
(18:31:05) (  2.6%) Carrier:   +93.8 Hz, Locked: No
(18:31:10) (  3.8%) Carrier: +1929.1 Hz, Locked: No
(18:31:15) (  5.1%) Carrier:  +140.4 Hz, Locked: No
(18:31:20) (  6.3%) Carrier:    +3.1 Hz, Locked: No
(18:31:25) (  7.5%) Carrier:    -3.2 Hz, Locked: No
(18:31:30) (  8.8%) Carrier:   +33.9 Hz, Locked: No
(18:31:35) (  9.6%) Carrier:   +35.4 Hz, Locked: No
(18:31:40) ( 10.3%) Carrier:   -19.5 Hz, Locked: No
(18:31:45) ( 11.0%) Carrier:    -8.1 Hz, Locked: No
(18:31:50) ( 11.8%) Carrier:    -0.6 Hz, Locked: No
(18:31:55) ( 12.9%) Carrier:   +44.7 Hz, Locked: No
(18:32:00) ( 13.8%) Carrier:   +13.4 Hz, Locked: No
(18:32:05) ( 14.6%) Carrier:   -15.3 Hz, Locked: No
(18:32:10) ( 15.3%) Carrier:    -9.3 Hz, Locked: No
(18:32:15) ( 16.6%) Carrier:  +301.5 Hz, Locked: No
(18:32:20) ( 17.8%) Carrier:   -86.4 Hz, Locked: No
(18:32:25) ( 19.0%) Carrier: -2728.6 Hz, Locked: No
(18:32:30) ( 19.8%) Carrier:  -374.2 Hz, Locked: No
(18:32:35) ( 21.0%) Carrier: -1245.9 Hz, Locked: No
(18:32:40) ( 21.9%) Carrier:  -532.6 Hz, Locked: No
(18:32:45) ( 23.1%) Carrier: -2484.5 Hz, Locked: No
(18:32:50) ( 23.6%) Carrier:  +161.4 Hz, Locked: No
(18:32:55) ( 24.7%) Carrier:  -488.9 Hz, Locked: No
(18:33:00) ( 25.6%) Carrier:   -49.1 Hz, Locked: No
(18:33:05) ( 26.2%) Carrier: +1955.7 Hz, Locked: No
[... 116 lines cut ...]
  File "/home/pi/raspberry-noaa/rectify.py", line 131, in <module>
    img = Image.open(sys.argv[1])
  File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2634, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/var/www/wx/images/METEOR-M220201223-181535-122-IR.png'
rm: cannot remove '/var/www/wx/images/METEOR-M220201223-181535-122-IR.png': No such file or directory
INFO : Rectifying image to adjust aspect ratio
convert-im6.q16: unable to open image `/var/www/wx/images/METEOR-M220201223-181535-122-IR-rectified.png': No such file or directory @ error/blob.c/OpenBlob/2874.
convert-im6.q16: no images defined `/var/www/wx/images/METEOR-M220201223-181535-122-IR-rectified.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert-im6.q16: unable to open image `/var/www/wx/images/METEOR-M220201223-181535-122-IR-rectified.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert-im6.q16: no images defined `/var/www/wx/images/thumb/METEOR-M220201223-181535-122-IR-rectified.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert-im6.q16: unable to open image `/var/www/wx/images/METEOR-M220201223-181535-122-IR-rectified.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert-im6.q16: no images defined `/var/www/wx/images/thumb/METEOR-M220201223-181535-122-rectified.jpg' @ error/convert.c/ConvertImageCommand/3258.
INFO : Posting to Twitter
Traceback (most recent call last):
  File "/home/pi/raspberry-noaa/post.py", line 23, in <module>
    res = api.media_upload(filename)
  File "/usr/local/lib/python3.7/dist-packages/tweepy/api.py", line 223, in media_upload
    file_type = imghdr.what(filename)
  File "/usr/lib/python3.7/imghdr.py", line 16, in what
    f = open(file, 'rb')
FileNotFoundError: [Errno 2] No such file or directory: '/var/www/wx/images/METEOR-M220201223-181535-122-IR-rectified.jpg'
rm: cannot remove '/var/www/wx/meteor/METEOR-M220201223-181535.bmp': No such file or directory

```

reynico, 2020-12-23:
Hi, Seems like the pass recording and decoding process did it fine, but the Meteor didn't output anything. Nothing to do from our side I guess

Anjum9694, 2020-12-24:
So how can I do a full clean uninstall so I can install it again

reynico, 2020-12-24:
You’ll need to remove the entire raspberry-noaa folder, .predict folder, the .noaa.conf, .wxtoimglic and .wxtoimgrc files.

mihajlo2003petkovic, 2020-12-25:
Meteor still not getting decoded. The last version that worked for me was v1.4. Could I try installing that version and then running `git fetch -pt && git pull` to update it to the current version?

 
[image]

mihajlo2003petkovic, 2020-12-26:
Hi! It seems that now I am succesfully able to receive Meteor M2! Thanks for your help!

closed 2020-12-26 by mihajlo2003petkovic