reynico/raspberry-noaa #92: ISS (ZARYA)" 145.8000 is not working from schedule  only METEOR-M 2 NOAA19,18,15 works well

v222008, 2020-12-26:
```
wget -qr http://www.celestrak.com/NORAD/elements/weather.txt -O "${NOAA_HOME}"/predict/weather.txt
wget -qr http://www.celestrak.com/NORAD/elements/amateur.txt -O "${NOAA_HOME}"/predict/amateur.txt
grep "NOAA 15" "${NOAA_HOME}"/predict/weather.txt -A 2 > "${NOAA_HOME}"/predict/weather.tle
grep "NOAA 18" "${NOAA_HOME}"/predict/weather.txt -A 2 >> "${NOAA_HOME}"/predict/weather.tle
grep "NOAA 19" "${NOAA_HOME}"/predict/weather.txt -A 2 >> "${NOAA_HOME}"/predict/weather.tle
grep "METEOR-M 2" "${NOAA_HOME}"/predict/weather.txt -A 2 >> "${NOAA_HOME}"/predict/weather.tle
grep "ZARYA" "${NOAA_HOME}"/predict/amateur.txt -A 2 > "${NOAA_HOME}"/predict/amateur.tle

#Remove all AT jobs
for i in $(atq | awk '{print $1}');do atrm "$i";done

#Schedule Satellite Passes:
"${NOAA_HOME}"/schedule_iss.sh "ISS (ZARYA)" 145.8000
"${NOAA_HOME}"/schedule_meteor.sh "METEOR-M 2" 137.1000
"${NOAA_HOME}"/schedule_sat.sh "NOAA 19" 137.1000
"${NOAA_HOME}"/schedule_sat.sh "NOAA 18" 137.9125
"${NOAA_HOME}"/schedule_sat.sh "NOAA 15" 137.6200
```

mihajlo2003petkovic, 2020-12-26:
Hi, I've had the same problem, just run `sudo mkdir -p /var/www/wx/iss` and change `receive_iss.sh` script to this:
```
#!/bin/sh
datetime=$(date +"%Y%m%d-%H%M%S")
timeout 660 /usr/local/bin/rtl_fm -M fm -f 145.8M -s 48k -g $GAIN -p 55 -E wav -E deemp -F 9 - | sox -t raw -e signed -c 1 -b 16 -r 48000 - /var/www/wx/iss/iss-${datetime}.wav rate 11025
```

mihajlo2003petkovic, 2020-12-26:
By the way, it seems that sound isn't properly recorded since every file is 44 bytes large and doesn't contain any sound data.

lyovav, 2020-12-26:
> By the way, it seems that sound isn't properly recorded since every file is 44 bytes large and doesn't contain any sound data.

i'm just add 
```
## import common lib
. "$HOME/.noaa.conf"
```
in receive_iss.sh   and recoring  file is ok

reynico, 2020-12-27:
Hi, receive_iss still needs a bit of work. 

Just as a reminder: do not run scripts with sudo, rather than that create the folders and setup the permissions for the pi user.

dinoj-bg, 2020-12-28:
Try to set ppm with correct one and set gain manual. It is OK for me when I change folder and permissions. BTW I run on Intel CPU platform and different user, but this do not matter.

reynico, 2020-12-31:
@v222008 there's a new branch `iss_support` that handles almost everything for ISS reception and decoding

```
cd /home/pi/
mv raspberry-noaa raspberry-noaa-bak
git clone https://github.com/reynico/raspberry-noaa.git -b iss_support
cp raspberry-noaa-bak/sun.py raspberry-noaa/sun.py
cp raspberry-noaa-bak/panel.db raspberry-noaa/panel.db
```

and then run the migration
```
cd raspberry-noaa/migrations/
./20201292-iss.sh
```

give a try and let me know how it goes

dinoj-bg, 2021-01-02:
Please correct line 15 of migration script /die=echo/ 
Thanks

reynico, 2021-01-02:
fixed d946296

reynico, 2021-01-03:
Guys I just merged `iss_support` into master. I'll close this issue

closed 2021-01-03 by reynico