reynico/raspberry-noaa #96: Not really an issue, just a question/request - HVC false color with/without map?

tegwilym, 2021-01-02:
It seems the new web interface only does 3 different processing methods of the NOAA audio.  
Is there a way to add in code that will process a basic false color HVC with and without the map?  Seems that all the defaults have the color overlay which gives it the cartoon look to the images.   I know I can do it manually with xWxtoimg, but not sure how I can add that option to the .sh code.  Something like this one. (Eastern US this morning from NOAA18)
Great work on the software, I really enjoy it, and used it recently to get some good ISS SSTV images.  :-)

Tom

[image]

reynico, 2021-01-02:
Hi Tom! Glad you liked the software.

These are the enhancements for daylight passes:
```
ZA MCIR MCIR-precip MSA MSA-precip HVC-precip HVCT-precip HVC HVCT
```

This depends on the setting called `SAT_MIN_ELEV` inside `/home/pi/.noaa.conf`. So if the satellite max elevation for that pass is above `SAT_MIN_ELEV`, all the enhancements are applied.

Regarding the map overlay behavior, check this line of receive.sh: https://github.com/reynico/raspberry-noaa/blob/master/receive.sh#L48
Remove the
```
-m "${NOAA_HOME}/map/${3}-map.png"
```
string and that's it. I'll add a toggle for the noaa.conf file soon

tegwilym, 2021-01-02:
Ok thanks!  I bet a lot of my questions and IR images from Meteor are all related to the low sun angles these days too.  It will get better as days get longer.  
I did find that line with all the daylight pass settings and did see the HVC was in there. 
I'll play around with the other settings and see what happens.  I just get a little lost in the code since all the .conf and .sh files seem to be interwoven in some way.  I guess that's why I didn't get too far with programming.  :-)
Keep up the great improvements with the software.  It's super easy to use which is great too. 

Tom

> Hi Tom! Glad you liked the software.
> 
> These are the enhancements for daylight passes:
> 
> ```
> ZA MCIR MCIR-precip MSA MSA-precip HVC-precip HVCT-precip HVC HVCT
> ```

closed 2021-01-02 by reynico