reynico/raspberry-noaa #99: [Error] pd120_decode_master build compilation failed

ltiisidii, 2021-01-03:
**Describe the bug**
pd120_decode_master build compilation failed: library match has Fortran sources but no Fortran compiler found

**Screenshots and/or logs**

[image]

**Server information (please complete the following information):**
 - OS: Debian GNU/Linux 10 (buster) (raspberry pi 4 8gb) (Linux raspberrypi 5.4.79-v8+ #1373 SMP PREEMPT Mon Nov 23 13:32:41 GMT 2020 aarch64 GNU/Linux) 
 - Python version 3.7.3
 - SDR device RTL-SDR v3

**Posible Solution**

sudo apt-get install gfortran

reynico, 2021-01-03:
hi! thanks for the report. Mind you please install just `libgfortran5` to see if that handles the problem?

ltiisidii, 2021-01-03:
Thanks to you Nico for motivating me to start playing with noaa and iss, I was able to solve it by executing: sudo apt-get install gfortran and the compilation worked.

reynico, 2021-01-03:
thank you! fix already on master branch.

closed 2021-01-03 by reynico