reynico/raspberry-noaa #100: Message at end of install when running install.sh

tegwilym, 2021-01-04:
I seemed to have messed up something last night with an update, so I removed all the files and tried a fresh install after a reboot.

Rather than going to the final setup of lat/long at the end of the install, I got this message at the end then it stopped.   Here are the last few lines of what is shown during install process.  Looks like a folder missing and not set in the PATH? 

Tom 

> Collecting numpy==1.19.4 (from -r requirements.txt (line 1))
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.19.4-cp37-cp37m-linux_armv7l.whl (10.5MB)
    100% |████████████████████████████████| 10.5MB 30kB/s 
Collecting Pillow==8.0.1 (from -r requirements.txt (line 2))
  Downloading https://www.piwheels.org/simple/pillow/Pillow-8.0.1-cp37-cp37m-linux_armv7l.whl (1.3MB)
    100% |████████████████████████████████| 1.3MB 235kB/s 
Collecting scipy==1.5.4 (from -r requirements.txt (line 3))
  Downloading https://www.piwheels.org/simple/scipy/scipy-1.5.4-cp37-cp37m-linux_armv7l.whl (51.0MB)
    100% |████████████████████████████████| 51.0MB 6.3kB/s 
Installing collected packages: numpy, Pillow, scipy
  The scripts f2py, f2py3 and f2py3.7 are installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed Pillow-8.0.1 numpy-1.19.4 scipy-1.5.4
cp: cannot stat '{demod.py,utils.py}': No such file or directory
pi@raspberrypi:~/raspberry-noaa $

pbriesch, 2021-01-04:
Same for me.

reynico, 2021-01-04:
Hi! Sorry about that. should be fixed now in master.

tegwilym, 2021-01-04:
Wiped out the files and ran install.sh.  All worked, now just waiting for a pass to see what happens. 

> Hi! Sorry about that. should be fixed now in master.

tegwilym, 2021-01-04:
Fixed!  Installed perfectly and I saw NOAA19 again.   Happy to help with letting you know if I find any strangeness with the software.  :-)

reynico, 2021-01-04:
thank you! I'll close this issue for now, feel free to re open it if needed.

closed 2021-01-04 by reynico