reynico/raspberry-noaa #102: NOAA MCIR and NOAA MCIR-precip identitical

dom1706, 2021-01-06:
Hello,
I would like first to thank you for your wonderful raspberry-noaa script !

I have two questions:
1. the noaa mcir and noaa mcir-precip are identical. Where could be my problem ?
2. is it an easy way to put passes and images on a webpage ?

Thanks in advance for your help.
Dominique,

reynico, 2021-01-08:
Hi dom!
1. That's weird! Should be different. Could you please upload them?
2. yeah, the setup has a webserver included if you installed it recently. access through your browser at http://your.raspberry.pi.ip/

dom1706, 2021-01-08:
[image]
[image]
Hello,
Thanks for your reply.
1. Here are two pictures captured yesterday (MCIR and MCIR-precip). You can so see they are identical.
2. Yes, I know I can see the webpage locally with the IP address of the PI. But I want to put passes and pictures on an external website.
Best regards.
Dominique,

dom1706, 2021-01-08:
Hello again,
I've received HVCT-precip pictures and (as MCIR), no colours => same picture as the HVCT one.
Here, you'll find the log ... maybe it could help to find the problem:

Date: Fri, 08 Jan 2021 10:36:24 +0100

INFO : Starting rtl_fm record
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 49.60 dB.
Tuned to 138392500 Hz.
Oversampling input by: 32x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Allocating 15 zero-copy buffers
Sampling at 1920000 S/s.
Output at 60000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Bulding pass map
Satellite: NOAA 18
Pass Start: 2021-01-08 09:18:40 UTC
Pass Duration: 15:38
Elevation: 51
Azimuth: 100
Direction: southbound
..INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
Gain: 14.2
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
Gain: 14.2
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
Gain: 14.2
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
Gain: 14.2
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
Gain: 14.2
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
Gain: 14.2
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
Gain: 14.2
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
Gain: 14.2
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
Satellite: NOAA 18
Status: signal processing............................
Gain: 14.2
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Deleting audio files
--------------------------------------------------------------------------------------

Hope you'll find the solution.
Best regards.
Dominique,

reynico, 2021-01-10:
> Here are two pictures captured yesterday (MCIR and MCIR-precip). You can so see they are identical.

Wow! It's been a while since the last time I saw that, but sometime happens. It's not related to the setup

> Yes, I know I can see the webpage locally with the IP address of the PI. But I want to put passes and pictures on an external website.

if you want to expose the webserver to the internet it's just a matter of setting up a port forward rule in your router to the Raspberry's IP and port 80

dom1706, 2021-01-18:
Hello reynico,
You can close this comment. I didn't see how I can do it myself.

I've installed again your script and I see now the precipitations in the MCIR-precip captures.

Thank you.
Dominique,
ON4DGF

closed 2021-01-18 by dom1706