reynico/raspberry-noaa #103: Bias tee and other settings

colinluthier, 2021-01-07:
I am trying to set up some of the basic features that were configured during install. I found the language finally. I notice the code has a BIAS_TEE variable but I can't seem to find where it is set. Also, it would be nice to have a location to change the Lat and Long for mobile stations.

I was thinking a quick script could show the variables and allow changes. Upon exiting it would modify the proper files.

Also, an option to show pass time in local time would be very helpful. 

Thank you! The install was super easy.

reynico, 2021-01-08:
Hi Colin!

Almost all the configuration is stored in `/home/pi/.noaa.conf`. There you can enable bias_tee setting it to `-T`:
```
BIAS_TEE="-T"
```

> Also, it would be nice to have a location to change the Lat and Long for mobile stations.

Yeah, I need to improve that setting. Right now you have to edit three files:
* `/home/pi/raspberry-noaa/sun.py`
* `/home/pi/.wxtoimgrc`
* `/home/pi/.predict/predict.qth`

> Also, an option to show pass time in local time would be very helpful.

Do you refer to the webserver? Please refer to https://github.com/reynico/raspberry-noaa/blob/master/WEBPANEL.md#satellite-pass-dates-are-wrong

colinluthier, 2021-01-08:
Getting the times set helps me plan. I did have the bias tee enabled but it only seems to work in two of the USB ports on the rPi4. Having some of the information in the wiki would be helpful. If I make a simple page from things I have found in the files may I post it? It would only be a start to a true wiki.

colinluthier, 2021-01-08:
Also, where is the call to wxtoimg so I can adjust the output files used? Thank you so much.

colinluthier, 2021-01-10:
I dug into this pretty deep the last few days. I have been getting blank images and the log (nano /var/mail/pi) is telling me that -T is an invalid option in rtl_fm and then does not turn on the radio. All the images are created blank.

<snip>
INFO : Starting rtl_fm record
/usr/local/bin/rtl_fm: invalid option -- 'T'
</snip>

Now, that I am writing this i notice the option is --T shown. 

The call in receive.sh is:
timeout "${6}" /usr/local/bin/rtl_fm ${BIAS_TEE} -f "${2}"M -s 60k -g$

I am not sure where the extra - is coming from. it is not in the .noaa.config file. Also, I can not find the -T option in the rtl_fm manual.
http://manpages.ubuntu.com/manpages/trusty/man1/rtl_fm.1.html#options. 

For now I am going to use:
rtl_biast -b 1
To force it on and have turned off the bias tee in the configuration. (https://github.com/rtlsdrblog/rtl_biast)

reynico, 2021-01-10:
Hi Colin!

I think the `--` is part of the invalid option message, not the BIAS tee param. Seems like the BIAS_TEE param is missing a `-`. Could you please post your .noaa.conf configuration file?

> Also, where is the call to wxtoimg so I can adjust the output files used? Thank you so much.

it's on `receive.sh`

colinluthier, 2021-01-10:
NOAA_HOME=/home/pi/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
RAMFS_AUDIO=/var/ramfs
SAT_MIN_ELEV=15
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=0
LOG_LEVEL=DEBUG
LAT=38.296
LON=-122.307
BIAS_TEE="-T"
DELETE_AUDIO="true"
FLIP_METEOR_IMG="true"
GAIN=50
SCHEDULE_ISS="false"

reynico, 2021-01-10:
Looks correct. The only thing I can think about is that the quotes globbing is doing something nasty over there. Could you please edit the line 34 on receive.sh? https://github.com/reynico/raspberry-noaa/blob/master/receive.sh#L34

and change `${BIAS_TEE}`  by `-T` So it end up like

```
timeout "${6}" /usr/local/bin/rtl_fm -T -f "${2}"M -s 60k -g $GAIN -E wav -E deemp -F 9 - | /usr/bin/sox -t raw -e signed -c 1 -b 16 -r 60000 - "${RAMFS_AUDIO}/audio/${3}.wav" rate 11025
```

colinluthier, 2021-01-10:
I tried adding some code to turn on the Bias Tee before the pass and shut it off after but the Bias Tee never powered up. I also tried your line of code and it gave me the same error:

Date: Sat, 09 Jan 2021 19:31:13 -0800

INFO : Starting rtl_fm record
/usr/local/bin/rtl_fm: invalid option -- 'T'
rtl_fm, a simple narrow band FM demodulator for RTL2832 based DVB-T receivers

Use:    rtl_fm -f freq [-options] [filename]

colinluthier, 2021-01-10:
I removed the LNA and turned off the Bias Tee and got good data and even a Meteor pass. Not nearly as clean of data as with the LNA. I'll keep trying to sort out the LNA.

reynico, 2021-01-11:
what if you run this line manually:

```
/usr/local/bin/rtl_fm -T -f  137.1M -s 60k -g 50-E wav -E deemp -F 9 - | /usr/bin/sox -t raw -e signed -c 1 -b 16 -r 60000 - "/tmp/test.wav" rate 11025
``

colinluthier, 2021-01-12:
That works! :/ 
I am not sure what is happening then. I have found out how to write the RTL-SDR EEPROM to keep the Bias Tee always on and that works like a charm but I would love to turn it off when not in use.

colinluthier, 2021-01-12:
I think I figured out wat was wrong. I didn't use a fresh SD card to install your scripts and my version of rtl_fm was missing some features, including the ability to write to the EEPROM. When I reinstalled it it fixed the flag problem. Sorry to waste your time.

closed 2021-01-12 by colinluthier