reynico/raspberry-noaa #105: [Error] sun.py - list index out of range

flewid, 2021-01-08:
Hey, 

Great program. Modifying it to work on 64 bit arm (yay!) but experiencing an issue with sun.py 

I am not sure why I'm getting this error with sun.py (my images are also showing up blank, but that's another story to work on later) - figured should started with the basics of getting the scripts to work without errors first. 

```
Location: Canada 
Lat: 45.963000
Long: -75.809160
Timezone: Eastern Time Zone (-5 GMT) 
```

**Here is my ~/.noaa.conf**

```
NOAA_HOME=/home/radio/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
RAMFS_AUDIO=/var/ramfs
SAT_MIN_ELEV=30
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=45.963000
LON=-75.809160
BIAS_TEE="enable_bias_tee"
DELETE_AUDIO="false"
FLIP_METEOR_IMG="true"
GAIN=42.0
SCHEDULE_ISS="true"
```

**Here is my ~/.prediect/predict.qth** 
```
BILLY
 45.963
 -75.8091
 223
```

**Here is my ~/.wxtoimgrc** 
```
Registration Name: Kevin Schuchmann
Registration Email: mail@mail.com
Registration Key: CGHZ-PP9G-EAJZ-AWKK-NDNX
Latitude: 45.963000
Longitude: -75.809160
Altitude: 228
```

**Here is my sun.py**
```
#!/usr/bin/env python3
import ephem
import time
import sys
timezone = 5 + time.localtime().tm_isdst
date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])-(timezone*60*60)))

obs=ephem.Observer()
obs.lat='45.963000'
obs.long='-75.809160'
obs.date = date

sun = ephem.Sun(obs)
sun.compute(obs)
sun_angle = float(sun.alt) * 57.2957795 # Rad to deg
print(int(sun_angle))
```

**And here is when I run sun.py** 
```
radio@radio:~/raspberry-noaa$ ./sun.py
Traceback (most recent call last):
  File "./sun.py", line 6, in <module>
    date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])-(timezone*60*60)))
IndexError: list index out of range
```

reynico, 2021-01-10:
Hi! sun.py expects an epoch formatted date/time string as the first argument so it calculates the sun elevation for that given time.

try:
```
./sun.py 1610239277
```

flewid, 2021-01-10:
Oh duh! Thanks. Maybe the individual scripts should all have a warning "Please run schedule.sh instead of this one" and hide the error output that's given when they are run?

tomamdy, 2021-01-10:
I see by flewid:
Here is my ~/.prediect/predict.qth

BILLY
 45.963
 -75.8091
 223

Here is my ~/.wxtoimgrc

Registration Name: Kevin Schuchmann
Registration Email: mail@mail.com
Registration Key: CGHZ-PP9G-EAJZ-AWKK-NDNX
Latitude: 45.963000
Longitude: -75.809160
Altitude: 228

But I thought that predict qth and wxtoimg there must be a difference with - for longtitude?

Tom

flewid, 2021-01-10:
You are correct. I changed

BILLY
45.963
-75.8091
223

to

BILLY
45.963
75.8091
223

and passes are predicted correctly and NOAA images come in properly :)

reynico, 2021-01-10:
Thanks Tom!

closed 2021-01-11 by reynico