reynico/raspberry-noaa #106: Meteor M2 daylight passes?  Sun / Satellite setting in .noaa.conf?

tegwilym, 2021-01-09:
First, thanks again for the excellent software!

Ok my question:   I know this time of year in the US is winter/darker for passes so we have a narrow window for daylight passes.  I do see there are Meteor M2 passes that should produce a daylight image (I do see them posted on the Weather Satellite Imaging Facebook group).  
I'm in NE Wisconsin, and do see images from another station south of me, and other daylight from a guy in Michigan.  I just keep getting IR images.   Is there a setting, sun angle, satellite angle or similar that I need to adjust to get daylight from Meteor?
Currently, this is what I have in my .noaa.conf file:

> NOAA_HOME=/home/pi/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
RAMFS_AUDIO=/var/ramfs
SAT_MIN_ELEV=18
METEOR_MIN_ELEV=10
SUN_MIN_ELEV=5
LOG_LEVEL=DEBUG
LAT=44.861526
LON=-87.349044
BIAS_TEE="-T"
#DELETE_AUDIO="true"
DELETE_AUDIO="false"
FLIP_METEOR_IMG="true"
GAIN=50
SCHEDULE_ISS="false"

reynico, 2021-01-10:
Hi! `SUN_MIN_ELEV` is the setting you are looking for. You can also set negative values there

tegwilym, 2021-01-10:
> Hi! `SUN_MIN_ELEV` is the setting you are looking for. You can also set negative values there

Oh, I didn't realize sun could go negative.  That make sense, I'll give it a try and see what happens.  Thanks!
-Tom

flewid, 2021-01-10:
How do we determine the optimal setting for SUN_MIN_ELEV I'm in a similar location to tegwilym and am also only getting IR images. I had it set at 10, but now I've put it down to 1 to test tomorrow mornings pass.

tegwilym, 2021-01-10:
> How do we determine the optimal setting for SUN_MIN_ELEV I'm in a similar location to tegwilym and am also only getting IR images. I had it set at 10, but now I've put it down to 1 to test tomorrow mornings pass.

As an experiment, I tried setting mine from 5 to -5. I’ll report my results. 
Tom

flewid, 2021-01-10:
> > How do we determine the optimal setting for SUN_MIN_ELEV I'm in a similar location to tegwilym and am also only getting IR images. I had it set at 10, but now I've put it down to 1 to test tomorrow mornings pass.
> 
> As an experiment, I tried setting mine from 5 to -5. I’ll report my results.
> Tom

Cool. I tried mine at 1 and it's still just getting the IR image. How did yours turn out?

tegwilym, 2021-01-10:
I had a pass at 9:00 am.   Still just an IR image on this end.  I had it set to -5 on sun angle.  Sounds like same result as yours. 

> Cool. I tried mine at 1 and it's still just getting the IR image. How did yours turn out?

flewid, 2021-01-11:
I tried setting mine to 50 today for the pass, and it is still coming in IR - not sure what else to try here.

flewid, 2021-01-11:
So, -5, 10, 50, 1 have been tried

tegwilym, 2021-01-11:
My -5 didn't work and someone on the Facebook group said to try 1, so I set that.  Somewhere in the files maybe there is an IR setting somewhere?  I'm sure one of the developers will jump in here with a new idea!

flewid, 2021-01-11:
I've been looking through the code to try and figure this out but haven't had much success yet. I saw on the old instructables article they have separate settings for 'winter vs summer' and i'm wondering if that's what's going on here?

reynico, 2021-01-11:
perhaps a bad setting in lat/lon for sun.py?  could you please post the content of the sun.py file?

flewid, 2021-01-11:
here's mine 

```
#!/usr/bin/env python3
import ephem
import time
import sys
timezone = 5 + time.localtime().tm_isdst
date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])-(timezone*60*60)))

obs=ephem.Observer()
obs.lat='45.963000'
obs.long='-75.809160'
obs.date = date

sun = ephem.Sun(obs)
sun.compute(obs)
sun_angle = float(sun.alt) * 57.2957795 # Rad to deg
print(int(sun_angle))
```

flewid, 2021-01-11:
my timezone is EST or, GMT -5 , so that 5 is correct eh?

tegwilym, 2021-01-11:
> perhaps a bad setting in lat/lon for sun.py? could you please post the content of the sun.py file?

Sure, here is mine.  I didn't change anything in this so it's still whatever the default from the Github is. 

> #!/usr/bin/env python3
import ephem
import time
import sys
timezone = 6 + time.localtime().tm_isdst
date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])-(timezone*60*60)))

obs=ephem.Observer()
obs.lat='44.861526'
obs.long='-87.349044'
obs.date = date

sun = ephem.Sun(obs)
sun.compute(obs)
sun_angle = float(sun.alt) * 57.2957795 # Rad to deg
print(int(sun_angle))
~

reynico, 2021-01-11:
> my timezone is EST or, GMT -5 , so that 5 is correct eh?

for sun.py it should be `-5` as your timezone!  try changing 
```
timezone = 5 + time.localtime().tm_isdst
```

to 

```
timezone = -5 + time.localtime().tm_isdst
```

I think that was your problem

tegwilym, 2021-01-11:
> my timezone is EST or, GMT -5 , so that 5 is correct eh?

Yep.  I'm in Wisconsin and -6 for non-daylight savings time, is what I've been using and works fine.

flewid, 2021-01-11:
oh interesting (i didn't use your installer script, due to being on an odroid) but i copied the code out, and ran it manually and it spit out 5 instead of -5

flewid, 2021-01-11:
changed to -5 now

reynico, 2021-01-11:
> oh interesting (i didn't use your installer script, due to being on an odroid) but i copied the code out, and ran it manually and it spit out 5 instead of -5

bash and numbers... what a headache! sorry about that guys, hope that's the issue

tegwilym, 2021-01-11:
> > oh interesting (i didn't use your installer script, due to being on an odroid) but i copied the code out, and ran it manually and it spit out 5 instead of -5
> 
> bash and numbers... what a headache! sorry about that guys, hope that's the issue

So are you saying I should change this "6 +"

> timezone = 6 + time.localtime().tm_isdst

To a "-6 +"  ?

reynico, 2021-01-11:
@tegwilym yeah, should be
```
timezone = -6 + time.localtime().tm_isdst

```

flewid, 2021-01-11:
> > oh interesting (i didn't use your installer script, due to being on an odroid) but i copied the code out, and ran it manually and it spit out 5 instead of -5
> 
> bash and numbers... what a headache! sorry about that guys, hope that's the issue

No prob! Bash is a pita with numbers :)

tegwilym, 2021-01-11:
> @tegwilym yeah, should be
> 
> ```
> timezone = -6 + time.localtime().tm_isdst
> ```

Changed!  I'll see what happens. I have been getting images fine from the passes that show up on the schedule, so this just affects the daylight ones?

reynico, 2021-01-11:
yeah it just switches the enhancements, using +6 and your lat/lon settings gives a negative sun elev for almost all the passes

tegwilym, 2021-01-12:
> yeah it just switches the enhancements, using +6 and your lat/lon settings gives a negative sun elev for almost all the passes

Ok, so the sun angle was getting confused by that wrong setting I get it.  I'll have to wait until tomorrow for daylight, but I'll let you know how it works on my end.   Great  work  on all the upgrades to the software.  I'm always happy to help beta test and report the bugs!   :-)

tegwilym, 2021-01-12:
This morning's results -- fixed! 
The missing negative brought back all the daylight images on both Meteor and NOAA, and the full group of all the processed images in NOAA.  I was only getting the first 3 IR variations, and now I get all the daylight also.  

They are all nicely processed now. 
[image]

Meteor is back in color!  Sunny morning in Chicago. 
[image]

reynico, 2021-01-12:
glad it worked!

closed 2021-01-12 by reynico

tegwilym, 2021-01-12:
Just the latest tweak to fix on the installer.   Works great!
Looking forward to the next features.   ;-)

flewid, 2021-01-17:
@reynico I'm still getting black and white for daytime passes? I got one color a few days ago, but generally they are all in IR 

here's a sample

[image]

flewid, 2021-01-17:
Here are my configs: 

**.noaa.conf**

```
NOAA_HOME=/home/radio/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
RAMFS_AUDIO=/var/ramfs
SAT_MIN_ELEV=30
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=45.963000
LON=-75.809160
BIAS_TEE="-T"
DELETE_AUDIO="false"
FLIP_METEOR_IMG="true"
GAIN=42.0
SCHEDULE_ISS="true"
```

**.predict/predict.qth**

```
BILLY
 45.9626
 75.8095
 223
```

**sun.py** 

```
#!/usr/bin/env python3
import ephem
import time
import sys
timezone = -5 + time.localtime().tm_isdst
date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])-(timezone*60*60)))

obs=ephem.Observer()
obs.lat='45.963000'
obs.long='-75.809160'
obs.date = date

sun = ephem.Sun(obs)
sun.compute(obs)
sun_angle = float(sun.alt) * 57.2957795 # Rad to deg
print(int(sun_angle))
```

flewid, 2021-01-17:
Also, my setup is a NooElec Smartee XTR + Sawbird NOAA and a National RF QFH antenna. Not sure if that makes a difference.