reynico/raspberry-noaa #107: Testing SDR Output from the RPi Headphones

colinluthier, 2021-01-09:
To play a radio station through the rPi headphones for testing (this example plays 97.3 FM):

rtl_fm -f 97.3e6 -M wbfm -s 200000 -r 48000 - | aplay -r 48000 -f S16_LE

The existing test instructions are hard to work in Windows without disabling the firewall. It was faster to play it on the rPi.

tegwilym, 2021-01-09:
I never did get that working through Pi to Linux over the network. But it's always received the satellites, so I didn't worry about it.

colinluthier, 2021-01-09:
I am getting blank images and haven't sorted it out so I needed to test the output. It took me a couple hours to sort out the command. I left it here in case I need it again. ;)

closed 2021-01-09 by colinluthier