reynico/raspberry-noaa #109: [Error] No output image

pineapplee85, 2021-01-12:
**Describe the bug**
Hi there,
I had to re-install on another Pi (2) and am not getting any output images. It just appears with a pink box or a missed pass.
All of the schedules passes seem to line up correctly.

Thanks!

**Screenshots and/or logs**
If applicable, add screenshots to help explain your problem. Logs are always welcome as well

http://www.filedropper.com/pi_1

**Server information (please complete the following information):**
 - OS: Raspbian
 - Python version Whatever is installed with newest install script.
 - SDR device Nooelec RTL-SDR with external LNA filter

reynico, 2021-01-12:
Hi! Try doing

```
echo  "GAIN=50" >> /home/pi/.noaa.conf
```

pineapplee85, 2021-01-12:
Hi there,

Thanks for your reply. Is this the gain setting for the SDR device? If so - This would not help me in my situation as when the LNA is in the feed line that amplifies the signal, thus the SDR gain just overworks the receiver for no reason.
Thanks

pineapplee85, 2021-01-12:
Re-installed the entire OS and script and now is working as it should. Not sure what that was all about but it's good now.
Thanks!

closed 2021-01-12 by pineapplee85