reynico/raspberry-noaa #110: Meteor M2 image quality under Win10 and Raspberry

tomamdy, 2021-01-12:
Why is it that Meteor M2 under win10 Sdrsharp, LRPT decoder and Medet64 gives a better image than a comparable recording under Raspberry? When I compare the images, the quality difference is quite big. If necessary, I can send two images as comparison material. I think this is such a shame because I think Reynico's software compilation is great.

reynico, 2021-01-15:
Hi Tom
Please upload both pictures to compare! Haven't use Windows for a while, so I don't have real comparison data

tomamdy, 2021-01-16:
First one  raspberry software, second one Sdrsharp, LRPT decoder and Medet64.
[image]
[image]

tomamdy, 2021-01-18:
May be this helps, this part is from pi:
Total:       1.767562
Processing:   1.763140
Correlation:  0.000000
Viterbi:      0.000000
ECC:          0.000000
Remainder:    0.004422
Packets:      425 / 425
Elapsed time: 00:00:49.248
convert: length and filesize do not match `/var/www/wx/images/METEOR-M220210118-085349-122.bmp' @ warning/bmp.c/ReadBMPImage/839.
INFO : Rectifying image to adjust aspect ratio
Opened 1568x424 image
Spawning process  1
Spawning process  2
Spawning process  3
Spawning process  4

reynico, 2021-01-18:
Hi tom,
yeah that's a huge difference, please follow this issue where guys are testing different settings to improve the reception: https://github.com/reynico/raspberry-noaa/issues/114

closed 2021-01-25 by tomamdy