reynico/raspberry-noaa #111: reception problem

fabiomarigliano, 2021-01-13:
hi, I installed everything according to the guide, but when I receive the satellites, I find myself in black and white images, very different from the ones I see in your twitter account, do you have any idea why? I receive the ZA MCIR and MICIR precip images.
[image]

fabiomarigliano, 2021-01-15:
I still get black and white images, does anyone have a problem with me? Thanks

[image]

reynico, 2021-01-15:
Hi Fabio!
Seems like the reception is a bit noisy over there.  WHat's your hardware setup and location?

fabiomarigliano, 2021-01-15:
Hi, I use raspberry Pi3 and QFH antenna, before with the old script I received color images even if a little noisy now only in black and white. Since I put the new script I can no longer receive the images as before. I enclose an example
[image]

fabiomarigliano, 2021-01-15:
My location Italy Europe

reynico, 2021-01-15:
Could you please post the contents of the following files?

- /home/pi/.noaa.conf
- /home/pi/.wxtoimgrc
- /home/pi/.predict/predict.qth
- /home/pi/raspberry-noaa/sun.py

fabiomarigliano, 2021-01-15:
NOAA_HOME=/home/pi/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
RAMFS_AUDIO=/var/ramfs
SAT_MIN_ELEV=30
METEOR_MIN_ELEV=30
SUN_MIN_ELEV=1
LOG_LEVEL=DEBUG
LAT=40.0113100
LON=-18.1623700
BIAS_TEE="-T"
DELETE_AUDIO="true"
FLIP_METEOR_IMG="true"
GAIN=70
SCHEDULE_ISS="false"

fabiomarigliano, 2021-01-15:
#
# WXtoImg configuration file written by WXtoImg.
# "man wxtoimg" or wxtoimg.html for details.
# WXtoImg version 2.11.2 beta
#
#
Registration Name: Kevin Schuchmann
Registration Email: mail@mail.com
Registration Key: CGHZ-PP9G-EAJZ-AWKK-NDNX
Signal Type: APT
Expert Mode: false
Status Info Size: 0
Large Pixmap Support: false
Prompt on Exit: true
Decode in Record: true
Scroll in Record: true
Direction: northbound
Sample Rate: 11025.00
#Resurs Sample Rate: 11025.0
#Meteor3 Sample Rate: 11025.0
#Meteor2 Sample Rate: 11025.0
#SICH1M Sample Rate: 11025.0
#SICH Sample Rate: 11025.0
#Okean Sample Rate: 11025.0
#NOAA Sample Rate: 11025.0
#Meteosat Sample Rate: 11025.0
#GOES Sample Rate: 11025.0
#GMS Sample Rate: 11025.0
#MTSAT Sample Rate: 11025.0
Set Sample Rate: false
Sharpen: 0.6
JPEG Quality: 90
AVI Codec: RGB
AVI Transparent: true
AVI Quality: 85
AVI Scale: 0.5000
Max Frames: 16
Frame Rate: 1.000
Image Format: JPEG
Anaglyph Image Format: png
Composite Image Format: png
Messages: normal
Resync: disabled
Noise Filter: 0
Noise Threshold: 44
Crop: disabled
Contrast: var
Illumination Compensation: none
Gamma: 1.40
Despeckle: 2.00
No Signal Fill: true
Audio Directory: /home/pi/wxtoimg/audio
Image Directory: /home/pi/wxtoimg/raw
Save Directory: /home/pi/wxtoimg/images
Map Directory: /home/pi/wxtoimg/maps
Thumbnail Directory: /home/pi/wxtoimg/thumbnails
Template for Audio: false
Filename Info: UTC YYYYMMDDHHMM
Temperature Units: C
Distance Units: km
Enhancement: -
Auto Image Template: %s-%f-%e
Auto Image Enhancements: "-" "contrast -a" "contrast -b" "NO" "HVCT" "MSA" "MCIR"
Record Type: A
Record Name Format: %Y%m%d%H%M
BMPs for DA: false
Record Bits: 16
Record Level: 25
Record Delay: 60
Record Rate: 11025
[... 181 lines cut ...]
Web Page Template: /usr/local/lib/wx/template1.html
Web Page Enhancement: MSA,MCIR,veg,Pristine
Web Composite Enhancement: 
Web Page Thumbnail Size: 20.0
Web Page Filename: index.html
Web Page FTP Publish: false
Web Page FTP Passive: true
Web Page FTP Host: myisp.someplace.net
Web Page FTP User: 
Web Page FTP Passphrase: 
Web Page FTP Directory: wxtoimg
Web Page Folder Publish: true
Web Page Folder: /home/pi/wxtoimg/web
Web Page Add All: false
Minimum Solar Elevation: 0.0
Exclude from Composites: 
Exclude from Web: 
Minimum Scan Lines: 0
Use MCIR if MSA fails: false
Kepler Host: www.celestrak.com
Kepler Login: 
Kepler Passphrase: 
Use Proxy: false
Proxy Host: 
Proxy Port: 8080

fabiomarigliano, 2021-01-15:
Fabio
 40.0113
 -18.1624
 117

fabiomarigliano, 2021-01-15:
#!/usr/bin/env python3
import ephem
import time
import sys

timezone= +1 + time.localtime().tm_isdst
date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])-(timezone*60*60)))

obs=ephem.Observer()
obs.lat='40.016'
obs.long='-18.1638'
obs.date = date

sun = ephem.Sun(obs)
sun.compute(obs)
sun_angle = float(sun.alt) * 57.2957795 # Rad to deg
print(int(sun_angle))

fabiomarigliano, 2021-01-15:
they are in order as you asked for them

reynico, 2021-01-16:
everything seems fine over there.
could yo please upload the output of one or two entries of `mail`?  those are the logs for the sat passes

fabiomarigliano, 2021-01-16:
To: pi@raspberrypi
Message-Id: <E1l0i8w-0005ZS-3o@raspberrypi>
From: pi@raspberrypi
Date: Sat, 16 Jan 2021 10:45:14 +0100

INFO : Starting rtl_fm record
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 49.60 dB.
Tuner error set to 1 ppm.
Tuned to 138392500 Hz.
Oversampling input by: 32x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Sampling at 1920000 S/s.
Output at 60000 Hz.
Signal caught, exiting!

User cancel, exiting...
INFO : Bulding pass map
wxmap: warning: could not find matching pass to build overlay map.
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png'
Satellite: NOAA
Status: signal processing............................
Gain: 14.6
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png'
Satellite: NOAA
Status: signal processing............................
Gain: 14.6
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png'
Satellite: NOAA
Status: signal processing............................
Gain: 14.6
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png'
Satellite: NOAA
Status: signal processing............................
Gain: 14.6
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
wxtoimg: warning: no specific NOAA satellite (using NOAA 12/14 series model)
wxtoimg: warning: No map to use for MSA enhancement
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png'
Satellite: NOAA
Status: signal processing............................
Gain: 14.6
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
wxtoimg: warning: no specific NOAA satellite (using NOAA 12/14 series model)
wxtoimg: warning: No map to use for MSA enhancement
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png'
Satellite: NOAA
Status: signal processing............................
Gain: 14.6
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png'
Satellite: NOAA
Status: signal processing............................
Gain: 14.6
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png'
Satellite: NOAA
Status: signal processing............................
Gain: 14.6
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png'
Satellite: NOAA
Status: signal processing............................
Gain: 14.6
Channel A: 1 (visible)
Channel B: 4 (thermal infrared)
Traceback (most recent call last):
  File "/home/pi/raspberry-noaa/post.py", line 23, in <module>
    res = api.media_upload(filename)
  File "/usr/local/lib/python3.7/dist-packages/tweepy/api.py", line 231, in media_upload
    file_type = imghdr.what(filename, h=h) or mimetypes.guess_type(filename)[0]
  File "/usr/lib/python3.7/imghdr.py", line 16, in what
    f = open(file, 'rb')
FileNotFoundError: [Errno 2] No such file or directory: '/var/www/wx/image//NOAA1820210116-102634-MCIR-precip.jpg'
rm: impossibile rimuovere '/home/pi/raspberry-noaa/map/NOAA1820210116-102634-map.png': File o directory non esistente
INFO : Posting to Twitter
INFO : Deleting audio files

reynico, 2021-01-16:
okay, 
seems like your `predict.qth` settings are wrong, should be

```
Fabio
40.0113
18.1624
117
```
(Lon negated)

fabiomarigliano, 2021-01-16:
ok i try to change i will tell you in the next step. Thanks

maxzega, 2021-01-17:
Fabio is ok now with no negative longitude?

fabiomarigliano, 2021-01-17:
> Fabio is ok now with no negative longitude?

 No I always have the same problems, tomorrow I format and reinstall everything

maxzega, 2021-01-17:
Ok Thanks for the info

fabiomarigliano, 2021-01-19:
I solved by installing again. thank you all for the support I close

closed 2021-01-19 by fabiomarigliano