reynico/raspberry-noaa #113: [Enhancement] MeteorM2 different enhancements

Crosswalkersam, 2021-01-15:
Hi,

would it be possible to add different enhancements like a thermal enhancement to the METEOR images?

reynico, 2021-01-15:
Hi! As far I know, there are no enhancements on METEOR images (or at least, not as the enhancements for the NOAA ones)

Crosswalkersam, 2021-01-15:
The Program "LRPT Image Processor" is able to do that. 
You just have to install the runtimes from here: https://www.satsignal.eu/software/runtime.html
And then you can run the program from here: https://www.satsignal.eu/software/LRPT-processor.html
I just have no idea if it was possible to automate this on the raspberry.
[image]

reynico, 2021-01-16:
Thanks! I'll dig into it but I think you'll have better luck reaching out artlav: https://github.com/artlav/meteor_decoder

colinluthier, 2021-01-22:
I will try this option:

“ Nice false color image is produced with -r 65 -g 65 -b 64”

colinluthier, 2021-01-23:
I was just looking through my log and my last pass at 7:25 AM Local gave this log entry:
INFO : I got a successful METEOR-M220210123-074510.dec file. Decoding APID 68
The sun should have been high enough to have tried to get the other 3 channels.

reynico, 2021-01-24:
Hi colin!

What's your timezone? Could you please post a copy of your sun.py file?

colinluthier, 2021-01-24:
I sorted this out last night. I never can sort out what time zone things want vs Day light savings so I just picked Los Angles. That output 8 into sun.py Last night I changed it to -8 and set up some logging to verify the proper sun angle was being generated and it works now. 

Thank you!