reynico/raspberry-noaa #114: Correct Predict longitude and gain settings

maxzega, 2021-01-17:
Hi, I wanted to ask how you can delete the steps scheduled by mistake. 
If in predict/predict.qth the longitude must be negative (I live near Rome) or positive and from what I understand by setting the gain to 48 or 50 in noaa.conf the automatic gain of the key is activated. The other values ​​to be set as an alternative are those that occur in response to the rtl_test? Float or integer?
Thanks a lot

reynico, 2021-01-17:
Hi!

> Hi, I wanted to ask how you can delete the steps scheduled by mistake.

List the passes with `atq` and then use `atrm <pass id>`  to delete it. Example:

```
pi@wx:~ $ atq
2080	Sun Jan 17 20:45:00 2021 a pi
2077	Sun Jan 17 22:24:00 2021 a pi
2075	Sun Jan 17 19:29:00 2021 a pi
2073	Sun Jan 17 20:40:00 2021 a pi
pi@wx:~ $ atrm 2080 # <- Deletes the Sun Jan 17 20:45:00 2021 pass
```

>  in predict/predict.qth the longitude must be negative (I live near Rome)

longitude values for rome should be positive

> from what I understand by setting the gain to 48 or 50 in noaa.conf the automatic gain of the key is activated

automatic gain control is activated if you don't define the `-g` parameter. and, from what i know, it's just an integer.

maxzega, 2021-01-17:
Hi reynico, thanks for the replies. 
As for the gain value to be entered in noaa.conf, is there a range (minimum and maximum)? the default value I saw is 50 but I also see who has even set it to 70 ... Within what range do you recommend me to make attempts? I use a self-built v dipole antenna. Thanks again. 

I also attach a meteor image this morning with gain at 25.
What could I do in your opinion to improve reception?
[image]

reynico, 2021-01-18:
Hi!

Supported gain values varies from hardware to hardware, run `rtl_test` to get the valid gain values for your receiver. Example:

```
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000002

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
```

Right now I'm playing a bit with the `rtl_fm` reception bandwidth for Meteor sats. Went down from 288k to 100k and seems to be a bit better (less noise perhaps?) so I recommend you doing the same.

maxzega, 2021-01-18:
> Right now I'm playing a bit with the rtl_fm reception bandwidth for Meteor sats. Went down from 288k to 100k and seems to be a bit better (less noise perhaps?) so I recommend you doing the same.

Thank you, i try the change for next pass.

Have a good day!

dom1706, 2021-01-18:
Hello,
I have exactly the same questions as maxzega.
But I'm not sure to have understood when reynico said "Right now I'm playing a bit with the rtl_fm reception bandwidth for Meteor sats"... "from 288k to 100k".
I don't know where I can play with the bandwith in the script ??

Other question about the gain: I have exactly the same valid gain values when rtl_test is activated.
So what value do I have to put in noaa.conf ??

Best regards.
Dominique,
ON4DGF

surfrod, 2021-01-18:
agreed... in the script I see rtl_fm -s288k ... sox -r288k, both "help text" specify this as the sample rate. Is that what you're referencing?

maxzega, 2021-01-18:
> Other question about the gain: I have exactly the same valid gain values when rtl_test is activated.
So what value do I have to put in noaa.conf ??

try different values... now i am set to 25.4 and is not bad...

> agreed... in the script I see rtl_fm -s288k ... sox -r288k, both "help text" specify this as the sample rate. Is that what you're referencing?
Yes, i have changed it and waiting for pass...

dom1706, 2021-01-18:
Hello maxzega,
I've put 25.4 as a gain value in noaa.conf ... crossing fingers ;-)
I've also changed the bandwidth in receive_meteor.sh in 100k from 288k and waiting the next pass in 30 min ...

Thanks again.

maxzega, 2021-01-18:
> Hello maxzega,
I've put 25.4 as a gain value in noaa.conf ... crossing fingers ;-)
I've also changed the bandwidth in receive_meteor.sh in 100k from 288k and waiting the next pass in 30 min ...

Thanks again.

Great, I did the same thing, we'll update soon :)

maxzega, 2021-01-18:
not ok yet for me....

[image]

dom1706, 2021-01-18:
Hello again, same here ! 
[image]

Do you think we have to put again the defaut value for bandwidth (288k) ??

maxzega, 2021-01-18:
For the next pass i'll try GAIN=29.7 and 144k

dom1706, 2021-01-18:
Good idea. We have to find a gain allowing us to have good captures for Meteor but also for NOAA !
Is it possible to know if the gain value (e.g. 29.7) will be good for Meteor but also for noaa captures ?

Another question: I have put the ppm correction (-p 32 for me) in receive.sh
Is it useful to put also this parameter in receive_meteor.sh ??

Have a good evening.

PS: I didn't take the time to thank reynico for his script !

surfrod, 2021-01-18:
I put the changes to gain into the receive_meteor.sh script alone, not changing the global value, allowing to experiment with Meteor without changing anything else.

I haven’t had a pass yet to validate, but am following here to see your results 👍

dom1706, 2021-01-18:
surfrod, it's an evidence but I did not think about it ;-)
Waiting for the last NOAA pass in 20min to see if it improves something.

I've put the passes and captures on a webpage; it's easier to follow from everywhere:
https://www.astrophoto.be/meteorologie

maxzega, 2021-01-18:
> I put the changes to gain into the receive_meteor.sh script alone, not changing the global value, allowing to experiment with Meteor without changing anything else.

Good idea! I am still trying to improve even for the noaa but managing the two types of satellite separately as gain is a great idea, maybe double gain could be inserted in the next updates of the script.

> I have put the ppm correction (-p 32 for me) in receive.sh
Is it useful to put also this parameter in receive_meteor.sh ??

Yes, you can try !!

reynico, 2021-01-18:
Hi guys!

I'm using `100k` and gain `50` and I'm pretty happy with the results: https://weather.reyni.co/detail.php?id=1523. not perfect but good enough

maxzega, 2021-01-19:
Last pass for me gain 29.7 and 144k sr

[image]

dom1706, 2021-01-19:
Great capture of Meteor-M2 !!
Are the parameters gain 29.7 and 144k sr good also for NOAA captures ??

Best regards.
Dominique,

tomamdy, 2021-01-20:
The best would be to input a fixed signal to the antenna input, so that we can set a Meteor M2 photo condition independent of receiving conditions. But how do you generate such a signal?

surfrod, 2021-01-20:
well the changes to sample rate (100k, then 144k) both degraded the signal. 1st time it was with a lower gain and 100k, but that resulted in a blank image altogether. Went back to gain=50 and rate of 144k, and there was more of an image, but not as good as my original settings, as bad as those are.

This is about as good a meteor image that I will get (attached)
[image]

tomamdy, 2021-01-20:
Here gain to 29.7 and timeout changed s 288k to 144k en-r 288k to 144k. Not a better result too.
[image]

mbruski, 2021-02-01:
Good Morning all,

In the Meteor receive script, I changed b/w from 288k to 120k for better results.  Also output rtl_fm to a raw file (no-extension) instead of pipe directly to sox.  Run sox on a separate command line using the raw file as input to produce resampled I/Q file.  I also added --norm to the sox command as suggested by someone in another forum.  Then in meteor_demod command added -O 8 -f 128 --pll-bw 400.  Images still aren't perfect but they are better.  I'm still playing with settings as well.

I also split rtl_fm and sox commands in the noaa receive script.  The --norm in sox seems to degrade the image so I don't use it for noaa.  @dom1706 - b/w for noaa should be between 40k and 44k depending on how much noise/interference you experience.  If you could apply doppler correction to rtl_fm in realtime (wishful thinking) you could set it to 38K.

As for gain, I find max gain (29.7 for my SDRv3 dongle) works best for Meteor and 25.4 seems to be optimum for NOAA.  I'm using a Tall-Narrow QFH with 85db FM notch filter and no pre-amp.

maxzega, 2021-02-01:
> Good Morning all,

In the Meteor receive script, I changed b/w from 288k to 120k for better results. Also output rtl_fm to a raw file (no-extension) instead of pipe directly to sox. Run sox on a separate command line using the raw file as input to produce resampled I/Q file. I also added --norm to the sox command as suggested by someone in another forum. Then in meteor_demod command added -O 8 -f 128 --pll-bw 400. Images still aren't perfect but they are better. I'm still playing with settings as well.

I also split rtl_fm and sox commands in the noaa receive script. The --norm in sox seems to degrade the image so I don't use it for noaa. @dom1706 - b/w for noaa should be between 40k and 44k depending on how much noise/interference you experience. If you could apply doppler correction to rtl_fm in realtime (wishful thinking) you could set it to 38K.

As for gain, I find max gain (29.7 for my SDRv3 dongle) works best for Meteor and 25.4 seems to be optimum for NOAA. I'm using a Tall-Narrow QFH with 85db FM notch filter and no pre-amp.

Thanks mbruski, can you post your modified receive.sh and receive_meteor.sh ?

I have added some  var in noaa.conf for separate management of gain on noaa and meteor and sampletate/bw also. 

GAIN_NOAA=28
BW_NOAA=60
SR_NOAA=60000
SCHEDULE_METEOR="true"
GAIN_METEOR=29.7
BW_METEOR=140
SR_METEOR=144

in receive.sh 

timeout "${6}" /usr/local/bin/rtl_fm ${BIAS_TEE} -f "${2}"M -s ${BW_NOAA}k -g ${GAIN_NOAA} -E wav -E deemp -F 9 - | /usr/bin/sox -t raw -e signed -c 1 -b 16 -r ${SR_NOAA} - "${RAMFS_AUDIO}/audio/${3}.wav" rate 11025

in receive_meteos.sh

timeout "${6}" /usr/local/bin/rtl_fm ${BIAS_TEE} -M raw -f "${2}"M -s ${BW_METEOR}k -g ${GAIN_METEOR} | sox -t raw -r ${SR_METEOR}k -c 2 -b 16 -e s - -t wav "${RAMFS_AUDIO}/audio/${3}.wav" rate 96k

mbruski, 2021-02-01:
Here are both of my scripts.  My code/script baseline is from mid December and I have already tweaked the noaa.conf file to include some things I wanted.  I also did some re-ordering of things in the scripts... nothing significant.

for receive.sh (i changed the filename suffix to .txt so github would accept it...)
[receive.txt](https://github.com/reynico/raspberry-noaa/files/5907453/receive.txt)

for receive_meteor.sh (ditto...)
[receive_meteor.txt](https://github.com/reynico/raspberry-noaa/files/5907457/receive_meteor.txt)

mbruski, 2021-02-02:
So the price I pay for being careless when I added some comments to the receiver_meteor.sh script before I posted is a syntax error on the meteor_demod command line.  The corrected line should be this:

	meteor_demod -O 8 -f 128 -m qpsk --pll-bw 400 -B -q "${METEOR_OUTPUT}/${3}.s" "${RAMFS_AUDIO}/${3}.wav"

My apologizes for the error.

tomamdy, 2021-02-02:
Mbruskj, I have tried your lines and here nothing worked. I didn't understand why.

mbruski, 2021-02-03:
My sincere apologies.   I'm not sure what happened here.   My NOAA captures today are looking really good but my first two meteor captures of the day aborted for script errors and one I did manage to complete this evening looked pretty awful.  I blame myself for not being more careful when I edited these things and I committed the worst sin by not backing them up before I did that.  I'll keep working the meteor script and hope to get that back in order soon.

tomamdy, 2021-02-04:
I am very interested in Your Noaa.conf file.

mbruski, 2021-02-04:
Yesterday was a complete bust as I couldn't decode a single packet from any of three meteor passes.  This morning I finally got an acceptable pass with the following script:

[receive_meteor.txt](https://github.com/reynico/raspberry-noaa/files/5927035/receive_meteor.txt)

Here is the image I captured.  I can post the raw I/Q file from rtl_fm is anyone wants to play it and see if they can improve on it.
[image]

I'm still disappointed with the results because I get better images (including overlays) with happysats windows solution using SDR#, LRPTDecoder and MeteorGIS.  I might look into using mlrpt by Neoklis Kyriazis since that is an all in one receiver/demod/decoder solution that supposedly can be run in place of rtl_fm/meteor_demod/met_arm.

mbruski, 2021-02-04:
noaa.conf

NOAA_HOME=/home/mosaic/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
RAMFS_AUDIO=/var/ramfs
SAT_MIN_ELEV=28
METEOR_MIN_ELEV=20
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=39
LON=-76
GRID=FM19
BIAS_TEE=""
DELETE_AUDIO="false"
FLIP_METEOR_IMG="true"
GAIN=48.0
M_GAIN=49.7
PPM=-2

dguardigli, 2021-03-15:
Did anyone solved with Meteor captures?