reynico/raspberry-noaa #116: Audio from ISS

dom1706, 2021-01-18:
Hello,

I would like to thank reynico for his script !

In 3 days, an ARISS contact will be audible in Europe on 145.800 MHz
Is it possible to record the wav from ISS with this script?
As it's not a SSTV contact, I'm asking if it's possible...

I've seen this on an issue 23 days ago:
"cd raspberry-noaa/migrations/
./20201292-iss.sh"
Do I have the migration even I've all reinstalled and installed the script again 3 days ago ?

Best regards.
Dominique,
ON4DGF

reynico, 2021-01-18:
Hi Dominique,
If you installed the script from the source 3 days ago you don't need to run the migration file. Just enable it on `/home/pi/.noaa.conf` 

```
SCHEDULE_ISS="true"
```

audio files will be stored on `/var/www/wx/iss/`

dom1706, 2021-01-19:
Hello reynico,
Thanks for your reply.

I had no directory iss in var/www/wx/
So I have created it. Do I have to chmod it ??

Have a good day.

dom1706, 2021-01-22:
Hello,
I would like to give you news from the test of receiving/recoding audio from ISS.
Yesterday evening, there was an ARISS contact between the astronaut Mike Hopkins and Maine Regional School Unit 21, Kennebunk, ME, USA.
I've just heared the astronaut during a few seconds when the ISS was at his max elevation.
Do you think it's because I have a 137.5MHz NOAA/APT Filtered Preamp and it doesn't allow me to have a good reception on 145.800 MHz ?
Have a good day.
Dominique,
ON4DGF

tegwilym, 2021-01-23:
i have wondered about this also.  I do have a 137 LNA on my QFH antenna.  When the ISS was doing the SSTV transmissions from the Ham station, I dd get get some ugly signals I disconnected my feedline from the QFH antenna and put it on my discone antenna which is tuned for a wide range of freqs.  I used it directly without the LNA.  
I let the software run with ISS enabled and recorded the audio pretty well! which I decoded into SSTV images.   I just had to forego the NOAA images for a few days since they looked worse.  Here is one of my better images. :-)
-Tom  KA7VIK
[image]

> Hello,
> I would like to give you news from the test of receiving/recoding audio from ISS.
> Yesterday evening, there was an ARISS contact between the astronaut Mike Hopkins and Maine Regional School [...]

dom1706, 2021-01-24:
Hello tegwilym,
Thanks for your reply.
It confirms what I supposed; taking another antenna to hear an ARISS conatct or obtain a SSTV image from the ISS.
Thank  you !
Dominique,

closed 2021-01-24 by dom1706

tegwilym, 2021-01-25:
I think that _should_ do it.  I need to try this again with swapping antennas and see if I can hear any astronauts talking to ground stations or even just listen to some ground-iss-ground repeater operations.   
Share your results if you get it working!  :-)
Tom

> Hello tegwilym,
> Thanks for your reply.
> It confirms what I supposed; taking another antenna to hear an ARISS conatct or obtain a SSTV image from the ISS.
> Thank you !
> Dominique,

dom1706, 2021-02-01:
Hello tegwilym,
Could you please tell me what's the frequency (other than 145.800 MHz ?) for ground-iss-ground repeater operations ??
Thanks in advance.
Dominique,

tegwilym, 2021-02-01:
> Hello tegwilym,
> Could you please tell me what's the frequency (other than 145.800 MHz ?) for ground-iss-ground repeater operations ??
> Thanks in advance.
> Dominique,

It looks like the main freq to listen to is 145.8.  I found this-
https://www.ariss.org/contact-the-iss.html