reynico/raspberry-noaa #118: Error in the pass list

fabiomarigliano, 2021-01-20:
Hi everyone, I regularly receive pictures of both the noaa and the Meteor. If I run the atq command I regularly see the correct steps with my time. The problem arises when I go to see the steps in the web browser, in the steps menu it shows me the list with the wrong steps, in particular they are steps with 4 hours less. Example if a satellite passes at 19PM on the web I see it at 15PM, but the system receives it regularly at 19PM it is only an error that I find in the request of the strings. Can anyone tell me what can be caused?
Thanks
[image]

reynico, 2021-01-20:
Hi Fabio, please refer to https://github.com/reynico/raspberry-noaa/blob/master/WEBPANEL.md#satellite-pass-dates-are-wrong

fabiomarigliano, 2021-01-20:
> Hi Fabio, please refer to https://github.com/reynico/raspberry-noaa/blob/master/WEBPANEL.md#satellite-pass-dates-are-wrong

ok thanks solved

closed 2021-01-20 by fabiomarigliano