reynico/raspberry-noaa #122: [Error] Rarely getting Meteor in Color

flewid, 2021-01-25:
For some reason in 2 weeks I have only received 2 color Meteor, they are almost always black and white. 

I've tried 100, 144, 248 for the samplerate from that other thread just to see if that made a difference (it did not). 

Here are my details: 

**Location:** 
`Eastern Time Zone `

**GPS according to google maps:** 
```
45.962608
-75.809464

```
**Dongle**
`NooElec SmarTee XTR `

**noaa.conf** 
```
NOAA_HOME=/home/radio/raspberry-noaa
NOAA_OUTPUT=/var/www/wx
METEOR_OUTPUT=/var/www/wx/meteor
RAMFS_AUDIO=/var/ramfs
SAT_MIN_ELEV=25
METEOR_MIN_ELEV=25
SUN_MIN_ELEV=10
LOG_LEVEL=DEBUG
LAT=45.962608
LON=-75.809464
BIAS_TEE="-T"
DELETE_AUDIO="false"
FLIP_METEOR_IMG="true"
GAIN=42.0
SCHEDULE_ISS="true"
```

**predict.qth**
```
BILLY
 45.9626
 75.8095
 223
```

**sun.py**
```
#!/usr/bin/env python3
import ephem
import time
import sys
timezone = -5 + time.localtime().tm_isdst
date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])-(timezone*60*60)))

obs=ephem.Observer()
obs.lat='45.963000'
obs.long='-75.809160'
obs.date = date

sun = ephem.Sun(obs)
sun.compute(obs)
sun_angle = float(sun.alt) * 57.2957795 # Rad to deg
print(int(sun_angle))
```

tegwilym, 2021-01-25:
You might try lowering your SUN_MIN_ELEV in .noaa.conf to something lower.  I have mine set at 1 and I'm getting at least one daylight image a day from Meteor.   I'm in NE Wisconsin, so I'm low on daylight here too.

flewid, 2021-01-25:
@tegwilym thanks, will try that

flewid, 2021-01-27:
thanks @tegwilym setting it to 1 has worked!

tegwilym, 2021-01-27:
> thanks @tegwilym setting it to 1 has worked!
Great!  I'm still learning the tricks of working out the different things in the configuration files.  Glad I've learned enough to give some tech support!   :-)

klikis, 2021-01-30:
mine is set to 10, in fact I get black and white images, do you recommend me to set it to 1?
[image]

tegwilym, 2021-01-30:
> mine is set to 10, in fact I get black and white images, do you recommend me to set it to 1?

Yes, try a lower sun angle.  You can even us a negative number I’ve heard. I have mine set to 1 and I’m getting color on both meteor and NOAA again. 
Here is the Great Lakes area from yesterday morning. 
Tom 

[image]

klikis, 2021-01-30:
ok thanks

klikis, 2021-02-02:
thanks......

[image]

Eric738, 2021-02-09:
your pic is nice..my last réception..:) 
[image]