reynico/raspberry-noaa #126: Sun.py

DeadlyByNite, 2021-01-28:
I've noticed that an unusual amount of Meteor images and NOAA sats were just black and white. Digging further, I noticed that for NOAA daytime passes, it's supposed to output 9 different enhanced images and 3 enhanced nighttime images, however I was getting the opposite. Only 3 images from each sat during the day, and 9 at night causing the last 4 images to be the duplicates since those sensors are usually not working at night. Also in my panel.db the table decoded_passes and column daylight_pass values are all backwards from what they are supposed to be. Debugging further, I found that the sun.py file is outputting a negative number during the day and a positive number at night. If in the $NOAA_HOME directory, typing "python3 sun.py $EPOCHSECONDS" confirms this.

surfrod, 2021-01-28:
ok, it's 6:03PM local, dark outside

doing that outputs:

pi@raspberrypi:~/raspberry-noaa $ python3 sun.py $EPOCHSECONDS
17

what should it be?

DeadlyByNite, 2021-01-28:
It should output -17 meaning the sun is 17 degrees below the horizon.

DeadlyByNite, 2021-01-28:
Actually looking closer, my timezone offset is -6 but line #5 of sun.py reads timezone = 6 ...
Making it a negative value, from 6 to -6 definitely improves the accuracy and fixes my problems, however it doesn't fix the problem for anyone who uses the install.sh file. If I have the time I'll look at that particular script and see if I can fix it accordingly.

DeadlyByNite, 2021-01-29:
I believe a fix to install.sh would be to fix line 304 from `sed -i -e "s/change_latitude/${lat}/g;s/change_longitude/${lon}/g;s/change_tz/$(echo  "$tzoffset * -1" | bc)/g" "sun.py"` to `sed -i -e "s/change_latitude/${lat}/g;s/change_longitude/${lon}/g;s/change_tz/${tzoffset}/g" "sun.py"`

Not sure if it's relevant or not, but inside of my predict.qth, I've got lat and lon both positive values, whereas in .noaa.conf and .wxtoimgrc it's set correctly. 

I haven't looked at enough code to see where the file predict.qth is used, but I know it gets tricky because the predict program and wxtoimg use opposite +/- values for lat & lon. My latitude is a negative value in wxtoimg whereas it's not in predict. So if the predict.qth file is **only** for setting lat and lon value to use with the predict application, then it's setup correctly and the only fix that is needed is line 304 inside of the install.sh script.