reynico/raspberry-noaa #130: [Error] raspbian download

Tom-Neverwinter, 2021-02-07:
**Describe the bug**
A clear and concise description of what the bug is.

https://www.raspberrypi.org/downloads/raspbian/

raspbian download is dead. rspbian version is not stated. 

**Screenshots and/or logs**
If applicable, add screenshots to help explain your problem. Logs are always welcome as well

**Server information (please complete the following information):**
 - OS: [e.g. Raspbian]
 - Python version [e.g. 3.7.3]
 - SDR device [e.g. RTL-SDR v3]

**Additional context**
Add any other context about the problem here.

pineapplee85, 2021-02-08:
This would be the issue of the raspberry pi foundation, not here - I would assume. 
Maybe try again soon or write a post on the forum?

Cirromulus, 2021-02-08:
Raspbian is discontinued, it is called Raspberry OS now [here](https://www.raspberrypi.org/software/).

Notice that the install script currently does not support the newer OS.
I fixed it in a fork, but there happened a lot as well, so can't PR right now.