reynico/raspberry-noaa #134: prune.sh appears to remove all images/thumbs

mbruski, 2021-02-14:
Running prune.sh appears to delete all images/thumbnails, not just the oldest 10.

Reviewing the script, I can see where img_path gets set but not IMG_NAME.   If IMG_NAME is not set, bash will treat it as a NULL string.  Thus in the "find" statements ${IMG_NAME}*.jpg is treated as *.jpg and all images get selected and deleted.

jekhokie, 2021-02-14:
Hey @mbruski - this is actually fixed in the v2 version of raspberry-noaa, in addition to adding some time-based pruning capability. You can check it out here: https://github.com/jekhokie/raspberry-noaa-v2

maxzega, 2021-02-14:
hi Jekhokie, can you post the example sqlite3 script line to delete a specific image from the tatabase? Thanks so much