reynico/raspberry-noaa #137: [Error] No image output + audio sounds weird

pineapplee85, 2021-03-11:
**Describe the bug**
Images have stopped being output & when I download a copy of the audio file the click sounds really fast. If I try a manual record I get a weird image too.
Was working fine and then just suddenly started.

**Screenshots and/or logs**
If applicable, add screenshots to help explain your problem. Logs are always welcome as well

[http://www.filedropper.com/error_12](url)

Server information
 - OS: Raspbian
 - Python version Unknown
 - SDR device Nooelec RTL-SDR SmarTee

jekhokie, 2021-03-11:
@pineapplee85 I'm not sure if Nico is still working actively on this, but you might consider upgrading to https://github.com/jekhokie/raspberry-noaa-v2 (an extension of his platform/framework).

pineapplee85, 2021-03-11:
@jekhokie Was thinking about that but it does not have posting to twitter which I really like

jekhokie, 2021-03-12:
@pineapplee85 I actually added Twitter back in recently - you can see the details for setting it up here: https://github.com/jekhokie/raspberry-noaa-v2/blob/master/docs/twitter_push.md

pineapplee85, 2021-03-12:
@jekhokie just going to install now. Thanks!

closed 2021-03-12 by pineapplee85