reynico/raspberry-noaa #141: [Error] Meteor only showing timestamp

lazerdave, 2021-03-23:
**Describe the bug**
All Meteor images are blank except for the information header.

**Screenshots and/or logs**
[image]

**Server information (please complete the following information):**
 - OS: Raspbian Buster
 - Python version 3.7.3
 - SDR device RTL-SDR v3

**Additional context**
Add any other context about the problem here.

jeffreypkelly, 2021-03-31:
I am having the same problem on the daytime passes.
I get the evening pass - but I get all white on the daytime pass.
Since there is no noise at all - it doesn't seem to be reading an actual wav file.
http://www.k2sdr.com
Got to be some configuration error.

jeffreypkelly, 2021-04-02:
OK found the problem. Look at #126.  I had to make negative my UTC offset to -4 (America/New_York) in the sun.py script.
Also be aware that M2 is in 1-2-3 mode so 68 (infrared) is not being sent.